ARM-software/CMSIS_6 #59: Unknown pragma when compiling with GCC

finchky, 2023-10-23:
I pulled down the latest dev drop today. 

It appears there is a `clang` pragma in `cmsis_gcc_m.h` (maybe others too) that generates a build error.
[image]

If I delete this line, then it builds fine

JonatanAntoni, 2023-10-24:
Thanks for reporting this. I'd assume GCC to ignore pragmas for other compilers. Which version of GCC do you use?

In this case, the pragma got added long time ago when the only way to use Clang was via GCC compatibility. Nowadays, we have separate headers for Clang and could (probably should) remove these pragmas from GCC headers.

Would you like to contribute the changes that via pull-request?

finchky, 2023-10-24:
We are using ARM GNU Toolchain 12.2. And are treating warnings as errors.

Sure, I can make a pull-request

closed 2023-10-25 by renesas-kyle-finch