ARM-software/CMSIS_6 #63: __ARM_FEATURE_CMSE definition

holden-zenith, 2023-10-24:
Where is this supposed to be defined? It pops a ton of -Wundef for me using gcc.

JonatanAntoni, 2023-10-26:
This is expected to be pre-defined by the compiler once `-mcmse` compiler flag is used.
Can you please be more specific about what version of GCC you are using and give at least examples where this gives you warnings?

holden-zenith, 2023-10-27:
So in all of the core_cmx.h files it is used, it has `#if defined(__ARM_FEATURE_CMSE) && (__ARM_FEATURE_CMSE == X)` however in all of the cmsis_x_m.h files in m-profile the usage isn't guarded with a `#if defined`.

GuentherMartin, 2023-10-30:
It is ok to use a define in the following way `#if (__ARM_FEATURE_CMSE == 3)`  without  a `#if defined` guard because the compiler will use value `0` for the macro if it is not defined.

ReinhardKeil, 2023-10-30:
Maybe it would be a good idea to use in the header file a construct that hints the usage of the `-mcmse` compiler flag.  For example:

```
#if (!defined(__ARM_FEATURE_CMSE))
#warning "CMSE extensions are not enabled. Use compiler option: -mcmse"
#endif

JonatanAntoni, 2023-10-31:
> Maybe it would be a good idea to use in the header file a construct that hints the usage of the `-mcmse` compiler flag. For example:

This won't work. The flag `-mcmse` is only used when targeting secure mode but not for non-secure mode. The define `__ARM_FEATURE_CMSE` is used to detect which mode we compile for.

> It is ok to use a define in the following way #if (__ARM_FEATURE_CMSE == 3) without a #if defined guard because the compiler will use value 0 for the macro if it is not defined.

While this is true, the [GNU C Preprocessor docs](https://gcc.gnu.org/onlinedocs/cpp/If.html) has the following:

> The -Wundef option causes GCC to warn whenever it encounters an identifier which is not a macro in an ‘#if’.

@holden-zenith, would you raise a contribution changing all the `#if` statements so that they don't cause you warnings?

ReinhardKeil, 2023-10-31:
Change it a bit. Would this cause a warning?
#if (defined(__ARM_FEATURE_CMSE) &&  (__ARM_FEATURE_CMSE == 3))

JonatanAntoni, 2023-11-27:
Relates to #79.

@holden-zenith, I think we can conclude that we should not rely on undefined macros implicitly expanding to 0 if used in an expression.

Can we close this one?

closed 2023-11-27 by holden-zenith