ARM-software/CMSIS_6 #88: What is the current status of the CMSIS_6 project?

redbilledpanda, 2023-11-23:
Hello

I wish to migrate some CMSIS-5 code to CMSIS-6 (mainly since we are moving from armcc to arm clang v6). What is the current status of this project? Is it in alpha/beta or something else? Can it be used in prod code?

JonatanAntoni, 2023-11-23:
Hi @redbilledpanda,

I'd consider CMSIS 6 to be in beta phase not ready for production, yet. We are in the final steps to have the first official release, soon.

If you want to give it a try, now would be a perfect point in time. CMSIS 6 introduces some breaking changes when migrating from CMSIS 5. See the latest [dev drop release](https://github.com/ARM-software/CMSIS_6/releases/tag/dev%2Fv6.0.0-dev53) as a starting point.

Please let us know about all findings you may have. This gives as the chance to get issues sorted and fixed before the upcoming release.

Thanks,
Jonatan

JonatanAntoni, 2024-01-19:
Hi @redbilledpanda,

In the meanwhile we published first CMSIS 6 release.
Please feel free to provide additional feedback by raising further issues.

closed 2024-01-19 by JonatanAntoni