ARM-software/CMSIS_6 #110: armv8-m register define doesn't include BPU

GuangMing34, 2023-12-18:
In CMSIS/Core/Include/core_m33.h or CMSIS/Core/Include/core_starmc1.h, DWT related register define has been implemented, but BPU not.

JonatanAntoni, 2023-12-18:
DWT got added to access cycle counter for example, which can be used as independent timer source for Event Recorder. BPU is considered to be relevant for debuggers, only. What features of BPU are you missing from application point of view?

GuangMing34, 2023-12-21:
Yes, some register are designed to read/write by debugger only. DWT are also debug component, cpu won't set it without debugger, it's no meaning. But there are still declared in register header file.

As BPU, some registers, FP_CTRL, FP_PIDx, FP_CIDxand other register with default values, can still be read by v8m cpu with main extension. there are some information for soc verification engineer to read & check.

JonatanAntoni, 2024-01-08:
Hi @GuangMing34,

Sure, I think you are just the first one ever looking for these additional registers. Hence they got never added. Please feel free to contribute whatever you need access to. Or, if you don't feel confident raising a pull-request, let us know set set of registers you would like to have access to.

GuangMing34, 2024-06-13:
Hi JonatanAntoni,

I have create a pull request(#174) for this issue.

Thanks~

JonatanAntoni, 2024-06-13:
Hi @GuangMing34, 

I have tagged @AlbertHuang-CPU who is our expert for Star-MC1 to review your contribution. Many thanks for raising this.

GuangMing34, 2024-07-11:
pull-request #174 already merged, this Issue can be closed.
Thanks @JonatanAntoni and @AlbertHuang-CPU .

closed 2024-07-11 by GuangMing34