ARM-software/CMSIS_6 #115: A bug on IRQHANDLER_T and weak functions?

truhy, 2024-01-01:
I have a project where I am trying to add a context pointer to the IRQ handler.  There is support for modifying the irq_ctrl_gic.c through the defines and weak functions:
#define IRQ_GIC_EXTERN_IRQ_TABLE
#define IRQHANDLER_T

I've modified the IRQHandler_t to a typedef struct, and also overridden the related weak functions.  But this gives compile errors. 
 It is because the weak functions are still compiled, but are overridden after.  The two weak functions IRQ_Handler() and IRQ_GetHandler() produces scalar type error.  I think the solution is to apply the same fix as applied on the IRQ_Initialize(), so adding in #ifndef:

```
/// The Interrupt Handler.
__WEAK void IRQ_Handler (void) {
  #ifndef IRQ_GIC_EXTERN_IRQ_TABLE
    IRQn_Type irqn = GIC_AcknowledgePending ();
    if (irqn < (IRQn_Type)IRQ_GIC_LINE_COUNT) {
      IRQTable[irqn]();
    }
    GIC_EndInterrupt (irqn);
  #endif
}

/// Get the registered interrupt handler.
__WEAK IRQHandler_t IRQ_GetHandler (IRQn_ID_t irqn) {
  IRQHandler_t h;
  #ifndef IRQ_GIC_EXTERN_IRQ_TABLE
    // Ignore CPUID field (software generated interrupts)
    irqn &= 0x3FFU;

    if ((irqn >= 0) && (irqn < (IRQn_ID_t)IRQ_GIC_LINE_COUNT)) {
      h = IRQTable[irqn];
    } else {
      h = (IRQHandler_t)0;
    }
  #endif
  return (h);
}
```

VladimirUmek, 2024-01-03:
Hi,
what you propose it one of the possible solutions. It would be beneficial to know why you need to extend the IRQHandler_t and how does the context structure look like. What benefits does the modified functionality bring?

truhy, 2024-01-03:
Hi, thanks for looking into it.  I have resolved my problem using a different approach, so don't really need the changes.

I am working on the Arm Cortex-A9 on Intel's Cyclone V SoC, so using a-profile, with Arm GNU Toolchain Version 13.2.Rel1
The reason I needed to extend the IRQHandler_t was because of trying to register a serial UART interrupt handler, coded for the 3rd party library [Intel SoC FPGA HWLIB](https://github.com/altera-opensource/intel-socfpga-hwlib/tree/master)

In their library they will only accept an IRQ handler with a context pointer, something like this:
void uart_irqhandler(uint32_t icciar, void *context);
So all of my handlers are of this form.

In CMSIS (by default), the IRQ_SetHandler(IRQn_ID_t irqn, IRQHandler_t handler) will only accept an IRQ handler, for example of this prototype:
void uart0_irqhandler(void);

So I was unable to register any of my handlers with context.  My context is simply of type void *, and I pass the pointer of the 3rd party library UART handle instance:
ALT_16550_HANDLE_t handle;

The benefit I can think of is if one wants to redefine IRQHandler_t, see CMSIS: a-profile/include/irq_ctrl.h:
```
#ifndef IRQHANDLER_T
#define IRQHANDLER_T
/// Interrupt handler data type
typedef void (*IRQHandler_t) (void);
#endif
```

If you redefine that to any other type, I think you will get also the data type compile error, so I am wondering how one would use that option?

JonatanAntoni, 2024-01-08:
Hi @truhy,

If I get your request right, the easiest and least intrusive way to inject the context for a certain interrupt would be to define some (static) trampoline function.

For example:
```
void* uart_context;

void uart_irqhandler_context(void* context);

void uart_irqhandler(void) {
    uart_irqhandler_context(uart_context);
}
```

At some point one need to connect/store the context pointer for each relevant interrupt of course. I'd expect the compiler to reduce the runtime overhead of above code to a minimum.

closed 2024-03-06 by JonatanAntoni