ARM-software/CMSIS_6 #121: const-qualify variable to suppress compiler warning

escherstair, 2024-01-10:
I get a warning/suggestion from the compiler because in the following function
https://github.com/ARM-software/CMSIS_5/blob/a75f01746df18bb5b929dfb8dc6c9407fac3a0f3/CMSIS/Core/Include/core_cm7.h#L2195-L2199

in the line

```C
  uint32_t *vectors = (uint32_t *)SCB->VTOR;
```

the pointer `vectors` can/should be defines as pointer-to-const

```C
  const uint32_t *vectors = (const uint32_t *)SCB->VTOR;
```

I think that the same change can be applied to all the other `core_cmXXX.h` files

JonatanAntoni, 2024-01-10:
Hi @escherstair,

Thanks for pointing this out. I moved your proposal to CMSIS 6 as this should be the place to make such enhancements.

I wonder if `uint32_t` is the correct type at all. Unfortunately, changing type to `void(*VECTOR_TABLE_Type)(void)` would be a breaking change.

ReinhardKeil, 2024-01-10:
@escherstair which compiler are you using.  SCB->VTOR can also point to a RAM based vector table. Is in this case (const uint32_t *) the right definition?

It may depend therefore depend on global compiler optimizations and other usage of SCB->VTOR in your application.

escherstair, 2024-01-10:
@ReinhardKeil the suggestion is only related to adding `const`.
The type was `uint32_t*` and I don't want to change it.
The function
```C
__STATIC_INLINE uint32_t __NVIC_GetVector(IRQn_Type IRQn)
{
  uint32_t *vectors = (uint32_t *)SCB->VTOR;
  return vectors[(int32_t)IRQn + NVIC_USER_IRQ_OFFSET];
}
```
doesn't change the value pointed by `*vectors`, but it returns it.
And so a `pointer-to-const` should be preferred.

I use AC6 with maximum warnings, combined with cppcheck + clang-tidy analysis.