ARM-software/CMSIS_6 #122: Backward incompatibilities in processor files vs CMSIS-Core (M) v5.x

vovamarch, 2024-01-16:
The processor header files for Cortex-M devices in CMSIS-Core v6.0.0 have a number of differences incompatible to CMSIS-Core 5.x.
 - Is there a list of all such incompatible changes?
 - Is it feasible for an application to keep compatibility to both v6 and v5 variants?

vovamarch, 2024-01-16:
The CMSIS-Core header files are updated to be aligned with the latest versions of Cortex-M Technical Reference Manuals (TRMs) . In some cases this leads to incompatibility issues, such as #118 (CoreDebug_Type no longer defined).

The list of incompatible changes is added to CMSIS-Core revision history as [Breaking changes in CMSIS-Core v6 header files](https://arm-software.github.io/CMSIS_6/main/Core/core_revisionHistory.html#core6_changes).

omarArm, 2024-01-17:
Hello,

I have created this header file with a set of definitions that work as patches for some changes between CMSIS@5.9.0 and CMSIS@6.0.0

It is important to note that I am using the header file as a pre-include in my projects, so please don't use it unless you are using CMSIS@6.0.0
[cmsis_5_to_6_patch.h.zip](https://github.com/ARM-software/CMSIS_6/files/13963157/cmsis_5_to_6_patch.h.zip)

The way to use it as follows:

1) Unzip the file into a directory you are familiar with <my_directory>
2) In the *.cproject.yml file of your project, under the "add_path:" keyword, add <my_directory>
`project:

  add-path:
    - $OutDir(Secure)$                       # add the path to the secure project's output directory
    -  <my_directory>                         #directory of any files you would like to include. `
3) In the *.cproject.yml file of your project, under the "misc:" keyword, add the pre-include command for the compiler you are using. Example:
`project:
  misc: 
    - for-compiler: AC6
      C-CPP:
        - --include cmsis_5_to_6_patch.h`

For more information on how to configure your *.cproject.yml file, please refer to [Yml-Input-Format.MD](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-Input-Format.md#project) It has a comprehensive explanation of how to write your own *.cproject.yml along with some examples

HTRamsey, 2024-03-14:
What do we do about the DWT Lock Access Register now that it is removed? Is the `DWT->LAR |= 0xC5ACCE55` part no longer needed?

GuentherMartin, 2024-03-21:
Hello @HTRamsey,
As far I understand then these registers are not always implemented.

If the registers are not implemented, then LSR register is RAZ and this also identifies the LAR register as not implemented.
If the registers are implemented and the user wants to access them, then he needs to create a pointer to the register’s address to write/read via this pointer the content of the registers.

JonatanAntoni, 2024-05-06:
Just fyi, we've decided to re-introduce deprecated `CoreDebug` symbols for backward compatibility reasons in #153.

closed 2024-05-06 by JonatanAntoni