ARM-software/CMSIS_6 #133: CMSIS API to obtain fine grained caching information

redbilledpanda, 2024-02-16:
Hello

I have been looking around CMSIS API to find out more fine grained information about caching for the MSMP0 (Arm Cortex M0, v6 arch) family of MCUs similar to what is available with Texas Instrument's DriverLib function ['DL_CORE_getInstructionConfig'](https://software-dl.ti.com/msp430/esd/MSPM0-SDK/1_20_00_05/docs/english/driverlib/mspm0c110x_api_guide/html/dl__core_8h.html#a391d1d42cdc53a20949bf472a64ebda9). 

As of now I couldn't find anything that will return instruction caching, prefetch, and literal cache configuration. I looked at "MPU functions for Armv6-M/v7-M" as well as Cache functions but nothing mapped to what I need. Is there something that I can use as a sort of a workaround or perhaps I'm missing something here?

JonatanAntoni, 2024-02-19:
Hi @redbilledpanda,

Did you already try to get in touch with the TI community?
The API functions you are referring to are not part of CMSIS.

The Arm Cortex-M0+ processor does not have a built-in cache. Hence, the CMSIS cache functions do not apply.

redbilledpanda, 2024-02-20:
Hello @JonatanAntoni 

Of course not but I was wondering if a driver for this family is in the works. Of course for now I will stick to the TI specified solution however would be great to use it via a unified API layer that lands into this driver.

JonatanAntoni, 2024-02-20:
A unification in this area is very unlikely to happen. As I said above, this is a very TI specific feature. Hence, I expect a TI specific API and implementation is required which probably won't match any other Cortex-M0+-based device from other vendors.

redbilledpanda, 2024-02-21:
Understood. Thanks a bunch

closed 2024-02-21 by redbilledpanda