ARM-software/CMSIS_6 #134: In CoreValidation - improper use of #ifdef's

phelter, 2024-02-20:
When creating CMake scripts to include the CoreValidation as a target, and adding in [CV_CML1Cache.c](https://github.com/ARM-software/CMSIS_6/blob/b20d76ae65fdccd36420c16d1c74958f68955d93/CMSIS/CoreValidation/Source/CV_CML1Cache.c#L21)

The `#ifdef __ICACHE_PRESENT` is improperly used - it should be `#if defined(__ICACHE_PRESENT) && (__ICACHE_PRESENT == 1)`

Same goes for `__DCACHE_PRESENT`

Please update.

JonatanAntoni, 2024-02-21:
Thanks for pointing this out. Please feel free to contribute required changes by raising a pull-request at any time.

JonatanAntoni, 2024-02-22:
Hi @phelter,

This issue should be solved by #136.
Please close this issue if the fix is appropriate.

Thanks.

closed 2024-02-29 by JonatanAntoni