ARM-software/CMSIS_6 #135: [BUG] Missing register definitions between CMSIS_5 and CMSIS_6 for core_cm4.h

phelter, 2024-02-21:
When migrating from CMSIS_5 to CMSIS_6 the following Identification registers in ITM are no longer defined:

CMSIS_5 core_cm4.h include: [ITM->PID* and CID*](https://github.com/ARM-software/CMSIS_5/blob/a75f01746df18bb5b929dfb8dc6c9407fac3a0f3/CMSIS/Core/Include/core_cm4.h#L847-L859).

CMSIS_6 core_cm4.h include: [https://github.com/ARM-software/CMSIS_6/blob/b20d76ae65fdccd36420c16d1c74958f68955d93/CMSIS/Core/Include/core_cm4.h#L843-L844)

Why is there a change in the definition of registers?

JonatanAntoni, 2024-02-21:
I think there were doubts about these registers to be properly supported. Can you give me some background about your requirements, please?

phelter, 2024-02-21:
I have a series of verification tests that confirmed all register accesses read their reset values.  These were listed as registers in CMSIS_5, but now not listed in CMSIS_6.  

I believe the reason for the ID registers PID/CID are there for the following reason:
> If there is ever a change to the hardware component of the ITM for one of the hardware core releases, these register values will change to reflect the ID version of the old or new ITM hardware component.  This may at some future time require a load or a selection of a different software/firmware path depending on these register values.

These registers won't typically be used in normal operation but I am currently using them to confirm that I am able to read unique values from that AHB/APB component and this confirms the addressing and hardware address demux selection logic is operational.  The reason for doing this is due to the fact that this particular implementation (a test chip) has no scan-insertion and I would like to confirm functionality of particular Stuck-At or other failures in the manufacturing process through functional testing of the device.

JonatanAntoni, 2024-02-23:
Hi @phelter,

I recapped the decision to remove such PID/CID registers and similar. 

Firstly, CMSIS-Core aims to support the most relevant use cases. Hence, we don't cover 100% of the core register set but only the ones most likely required in applications. Putting too much might easily clutter the header files.

Secondly, we never had component identification consistently supported. I.e., in some components we had them but in others we missed them. For cleanup from CMSIS 5 to 6 we could have added them everywhere. But as we think component identification is a very rare edge case we decided to rather remove them.

I appreciate that this decision requires some people like you to provide component identification, manually. Perhaps we can have a separate header file just providing access to various component identification features? This header file could be shared as some CMSIS-Core AddOn only for user with a similar use case.

closed 2024-05-06 by JonatanAntoni