ARM-software/CMSIS_6 #139: clang system header in gcc headers

amir-skn, 2024-02-29:
Hi,

I am unable to compile due to a warning saying:
`Include/cmsis_gcc.h:28: error: ignoring '#pragma clang system_header' [-Werror=unknown-pragmas]`

Which is being caused by this bit:
https://github.com/ARM-software/CMSIS_6/blob/4d18b8de66ce12e38c3d933a9e5fbe642b71763d/CMSIS/Core/Include/cmsis_gcc.h#L28

Library and toolchain:  **CMSIS_6 Core library** and **arm-none-eabi 13.2.1** 

I think it should be changed to:
`#pragma GCC system_header   /* treat file as system include file */` 

Could you please check?

Regards
Amir

JonatanAntoni, 2024-02-29:
Hi @amir-skn,

Thanks for pointing this out. This indeed is an oversight.
cc @Masmiseim36

Masmiseim36, 2024-02-29:
Hello @amir-skn and @JonatanAntoni ,

thanks for the hint, I have created a Pull-Request for this: https://github.com/ARM-software/CMSIS_6/pull/141

@JonatanAntoni any chance to make this visible in the test suite?
regards

JonatanAntoni, 2024-03-06:
See #140. I added additional warnings for GCC.

closed 2024-03-06 by JonatanAntoni