ARM-software/CMSIS_6 #143: CMSIS-Driver USART events expansion

MaxFlatline, 2024-03-11:
Hello!
I think there is a need to add additional event in driver, "half buffer transfer" event. In NMEA applications it could be extremely helpful, to start parcing while dma continues to receive data. But it seems that currently you get only rx/tx/transfer_complete event. 
Thank you!

JonatanAntoni, 2024-03-11:
Hi @MaxFlatline,

Thanks for getting in touch.

Could you please check if double (multiple) buffering [[1](https://en.wikipedia.org/wiki/Multiple_buffering)] can solve your issue? I.e., instead of asking for a "half buffer transfer" event, try to split the buffer into two buffers of the half size. Once the first buffer is full (transfer complete event), instantly start another transfer for the other buffer. While the second buffer is used for transfer, you can process the first one without interfering.

Cheers,
Jonatan

[1] https://en.wikipedia.org/wiki/Multiple_buffering

MaxFlatline, 2024-03-11:
Hello @JonatanAntoni 
Thank youfor your reply. Yes, i can and probably will use it, though it will be less optimal solution, since in my mcu there is an hardware feature of half-complete transfer detection. For now i guess it well be sufficient, but on high speeds it'll certainly cause data corruption on reception

JonatanAntoni, 2024-03-11:
Hi @MaxFlatline,

Well, thanks for adding the important bit that your hardware has hardware half-complete transfer detection. As this seems to be an optional hardware feature we'd need to check how to add optional support for such events.

Can you please provide more details about the hardware you are using? Please be aware that just enhancing the CMSIS-Driver API by another (optional) event doesn't change much. In a second step the implementation for you device would need to be enhanced as well. It depends on the maintainer of the DFP to do so. Well, you could provide your own custom implementation of course.

Thanks,
Jonatan

MaxFlatline, 2024-03-11:
@JonatanAntoni 
Of cource, but it seems i'm a bit wrong, half-transfer detection is a feature of DMA, not USART module. I'm using an STM32L072 and it's DMA supports an interrupt on half-transfer event, so if you are using USART with DMA, you are able to use half-complete callback (in HAL)

MiloradCvjetkovic, 2024-03-19:
Hi @MaxFlatline, 
it looks like this feature is not common across different vendors.
In your particular case, that functionality does not have to be provided by the CMSIS-Driver but you can implement those callbacks (for example `HAL_UART_RxHalfCpltCallback`) with your custom functionality, for example in that callback you can send event to your thread that can start processing available data.