ARM-software/CMSIS_6 #159: Differentiating between CM33 and CM85 with preprocessor

renesas-kyle-finch, 2024-05-07:
In a previous issue submitted to CMSIS 5, I was told that it was no longer advisable to use the arch defines like `__ARM_ARCH_8M_MAIN__`  and to instead use `__ARM_ARCH`.

But dumping preprocessor defines for CM33 and CM85 both yield the same `__ARM_ARCH*` defines:

```
// Using GCC 13.2
#define __ARM_ARCH 8
#define __ARM_ARCH_8M_MAIN__ 1
#define __ARM_ARCH_EXT_IDIV__ 1
#define __ARM_ARCH_ISA_THUMB 2
#define __ARM_ARCH_PROFILE 77
```

Note that GCC generates `__ARM_ARCH_8M_MAIN__` for both CM33 and CM85. 

LLVM 17 generates the same as GCC, with the exception that it generates `__ARM_ARCH_8_1M_MAIN__` for CM85.

How should CM33 and CM85 be differentiated with the preprocessor?

JonatanAntoni, 2024-05-08:
Hi @renesas-kyle-finch, 

this seems still to be a shortcoming in GCC. According to ACLE one should be able to detect the architecture by looking to `__ARM_ARCH`, see https://developer.arm.com/documentation/101028/0012/5--Feature-test-macros.

I'll pass this on to our GCC experts.

renesas-kyle-finch, 2024-05-08:
A follow up, it seems like IAR may also be producing conflicting macros. It seems for CM85 it produces `__ARM_ARCH 801` but also produces `__ARM_ARCH_8M_MAIN__`

I see there is on open pull request that looks like maybe it is adding support for IAR, is this the case?
https://github.com/ARM-software/CMSIS_6/pull/92

ReinhardKeil, 2024-05-10:
CMSIS contains the revision number of each core using these defines:
https://arm-software.github.io/CMSIS_6/v6.0.0/Core/group__device__config.html

Even when a device vendor does not provide the exact revision of the implementation, the fallback ensures that this define exists: 
- https://github.com/ARM-software/CMSIS_6/blob/main/CMSIS/Core/Include/core_cm85.h#L222
- https://github.com/ARM-software/CMSIS_5/blob/master/CMSIS/Core/Include/core_cm85.h#L201

This information is therefore less dependent on Compiler internal defines specified by [ACLE](https://github.com/ARM-software/acle/releases/tag/r2024Q1) and maybe easier to query.

```
#if defined (__CM85_REV)    // running on Cortex-M85
 ...
```

I hope this helps.

renesas-kyle-finch, 2024-05-10:
I will give that a try.

As far as looking to the future, are the `__ARM_ARCH` and `__ARM_ARCH_ISA_THUMB` macros still the advised macros to use and what the compilers should be striving to produce?