ARM-software/CMSIS_6 #171: missing defined (__ARM_FEATURE_CMSE) in some cases

CanastraRF, 2024-06-04:
Hi
most check for __ARM_FEATURE_CMSE are like this
defined (__ARM_FEATURE_CMSE) && (__ARM_FEATURE_CMSE == 3U)
but in some lines are only 
(__ARM_FEATURE_CMSE < 3U)
so my tools reporting, that __ARM_FEATURE_CMSE can be undefined and interpreting as 0

please add everywhere a defined (__ARM_FEATURE_CMSE)

compare cmsis_armclang.h with cmsis_gcc.h

JonatanAntoni, 2024-06-05:
Hi @CanastraRF, thanks for raising this. Would you be able to provide the required changes as a pull-request to speed up things?

GuentherMartin, 2024-06-20:
Hello @CanastraRF, can you please check PR #177 if this solves your issue?

CanastraRF, 2024-06-20:
Hello 
I've checked PR https://github.com/ARM-software/CMSIS_6/pull/177 and for us it is a good solution