ARM-software/CMSIS_6 #191: Newbie Example CMSIS drivers code project / link procedures in MCUExpresso?

RKnappPMT, 2024-09-18:
Hi, I've written reams of C code over the years, but am a painful NEWBIE in your area of expertise; please help!  

I have been told that I have to get CMSIS drivers working on a new board which has the same CPU as an MIMXRT1060_EVKB (actually it's a 1062) -- get Hello World loopbacks working on its UARTS, ethernet, etc. using MCUXpresso's IDE and a .mex file for the new board he sent me.  

OK, so I have that new board but I ALSO have an NXP demo board which is running evkbmimxrt1060_hello_world -- just fine, import, build, debug, blink-blink -- on MCUExpresso, but now what?  I cannot see how to get these CMSIS drivers, which appear to be part of my workspace ready to go, linked into either the demo or our new board vs the s. the "native" (fsl?) drivers.  I cannot find any example of how this would be done; could anyone please point me to any boneheaded example using CMSIS for UARTS / ethernet / etc.?  

I know this isn't really a CMSIS_6 "issue," so apologies to all -- but it's sure a big issue for me!  Thanks, ~~ Robert

jkrech, 2024-09-18:
Please visit https://keil.arm.com, select "Boards" from the Hardware drop down at the top of the page.
Then enter the board name "MIMXRT1060-EVKB". On the page that opens you can search for examples.
I used "cmsis uart"
https://www.keil.arm.com/boards/nxp-mimxrt1060-evkb-e45a3d6/projects/?q=cmsis%20uart
Use the "Get Project" drop down button at the right hand side. This is a project that is using the CMSIS Solution project format. Note that this is a project  from NXP that is distributed via a CMSIS-Pack (see Pack Download" button). To get this project built, you can download and install the [CMSIS-Toolbox CLI](https://artifacts.tools.arm.com/cmsis-toolbox/) or more conveniently install VSCode and the MCUXpresso for VS Code from NXP or Keil Studio Pack from Arm.

RKnappPMT, 2024-09-18:
Jkrech, thanks for your speedy response!  Very helpful -- I've been avoiding VSCode but will have to reconsider as it sometimes does make more sense than wandering around Apache archives... (having spent a lot of time in rabbit holes, getting lost.)  So I'll try to go that direction.  THANKS, ~~ Robert