ARM-software/CMSIS_6 #192: About Event Recorder

Magictotal10, 2024-09-24:
I use the CMSIS 6.1 software package, Fusa6.1 compiler, my C++17, rtx5 rtos for my project (ARM-M4 core). I do not use MicroLib, and I set up my heap size as 0. If I open Event Recorder, my program could not run into main, and failed after becase of _sys_exit(it seems to be after calling _platform_post_stackheap_init). If I dont use Event Recorder, or set some heap size using Event Recorder, my program could successfully run.
Note: I have already retarget stdio、_sys_exit、_ttywrch， not using semihosting.

So, Event Recorder must be using the heap memory?

Could you solve my puzzle, appreciation.

chentang16, 2024-09-24:
I don't think Event Recorder will use heap. 
Since you are using C++ lib, you may check these manual pages 
- https://developer.arm.com/documentation/100073/0622/The-Arm-C-and-C---Libraries/Stack-and-heap-memory-allocation-and-the-Arm-C-and-C---libraries/Library-heap-usage-requirements-of-the-Arm-C-and-C---libraries
- https://developer.arm.com/documentation/100073/0622/The-Arm-C-and-C---Libraries/Stack-and-heap-memory-allocation-and-the-Arm-C-and-C---libraries/Avoiding-the-heap-and-heap-using-library-functions-supplied-by-Arm

andbararm, 2024-09-24:
Do you with the Event Recorder also enable/disable stdio output in your application? I am not aware that the Event Recorder uses heap. But io streams certainly do.

RobertRostohar, 2024-09-24:
Event Recorder does not use heap. When using stdio, at least some heap is required and setting it to 0 would normally fail.

It would be good to understand in more detail what you are doing, maybe a call sequence, preferably having a simple reproducer project.

Magictotal10, 2024-09-25:
> I don't think Event Recorder will use heap. Since you are using C++ lib, you may check these manual pages
>
> * [...]

Thanks a lot. 
These two articles are very useful.
I actually found my code is implicitly using malloc and free function through command ‘__asm(“.global __use_no_heap_region”)’.

closed 2024-09-25 by jkrech