ARM-software/CMSIS_6 #194: Structs for memory mapped core register definitions

tudortimi, 2024-09-28:
Are there any plans to also define structs for the memory mapped core registers? Currently, only bitmasking is supported. Structs would make code easier to read (at the cost of more instructions, I guess).

For example, the SAU registers only have mask/pos defines:

[image]

For the non-memory mapped SFRs, there are also structs (e.g. APSR) in addition to the mask/pos defines:

[image]

JonatanAntoni, 2024-09-30:
Hi @tudortimi,

We do not have immediate plans to enhance CMSIS definitions.
Please feel free to raise a suggestion as a Pull-request.

I am happy to accept your contributions.

Thanks,
Jonatan

JonatanAntoni, 2024-10-01:
No, we are not using any tools. The CMSIS core header files are manually maintained so far.

tudortimi, 2024-10-02:
I'll raise a PR to add this fro one of the cores (e.g. CM33) first. It's gonna take some time, though, do you mind that the issue stays open? Also, could you assign it to me so it shows up in my TODO list?