ARM-software/CMSIS_6 #198: Why not use the compatible type __attribute__((__noreturn__)) as __NO_RETURN for IAR?

SuperHeroAbner, 2024-11-06:
The type of __attribute__((__noreturn__)) can be put before or after function like 'void func(void) __NO_RETURN;'. If use the type of [[noreturn]] or _Noreturn, it must be like '__NO_RETURN void func(void)'
https://github.com/ARM-software/CMSIS_6/blob/main/CMSIS/Core/Include/cmsis_iccarm.h#L57
This introduces compatibility issue. Does __attribute__((__noreturn__)) have any issue to apply on IAR compared with _Noreturn + [[noreturn]] ?

JonatanAntoni, 2024-11-06:
The compiler header for IAR is provided and maintained by IAR. Please consider raising this question directly with them as I am not aware how frequently they are watching CMSIS issues.

ReinhardKeil, 2024-11-06:
@jlonnberg maybe you can help.

RobinKastberg, 2025-01-22:
@SuperHeroAbner 
We reasoned that it was better to follow C standard and we had an issue from a customer that wanted to allow usage of __NO_RETURN in strict language mode.

We can try to have another go here but it is hard to please everyone.

What are the benefits of having __NO_RETURN afterwards and what compatibility did we break? I recall something in trustedfirmware-m?

SuperHeroAbner, 2025-02-06:
@RobinKastberg I just find the new type in IAR gets compiling issue if code is like 'void func(void) __NO_RETURN;', previously it's OK. But if the new way is better for certain 'strict language mode', yes we should use better one. And maybe it should be put in front of function as a law.