ARM-software/CMSIS_6 #200: Different #include "RTE_Components.h" situation

SuperHeroAbner, 2024-11-15:
There're different types of  #include "RTE_Components.h". What're the use cases for them?
1. No mask, [example](https://github.com/ARM-software/CMSIS_6/blob/main/CMSIS/RTOS2/Source/os_systick.c#L28C1-L28C28).
2. Mask as not define RTE_COMPONENTS_H, [example](https://github.com/ARM-software/CMSIS-RTX/blob/main/Source/rtx_core_cm.h#L30).
3. Mask as define RTE, [exmaple](https://github.com/ARM-software/CMSIS-RTX/blob/main/Config/RTX_Config.h#L31).

I just know Keil package input takes RTE macro definition + generated RTE_Components.h to implement point 3. Why we still need the RTE_COMPONENTS_H to mask at other place or force to include this file.

ReinhardKeil, 2024-11-15:
The `rte_compontents.h` is described here https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-overview.md#rte_componentsh

Build systems that maintain the `rte_compontents.h` file also provide the define symbol `_RTE_`. (This information will get added to the documentation).

Above examples are different usages:
1. Assumes that it is always used in an `_RTE_` enabled system.
2. SW has an alternative way to supply the `CMSIS_device_header`
3. The proposed way to use  `rte_compontents.h`.

I hope this helps.

JonatanAntoni, 2024-11-18:
Hi @SuperHeroAbner,

I guess the confusion comes from not being consistent across various software components.

We consider an "RTE enabled system" to implement the CMSIS-Pack approach. I.e., as a user one doesn't (manually) add files to the build but leave it to the "RTE resolver" to pick the required files. E.g., this is implemented in Keil uVision or CMSIS-Build.

These tools handle runtime configuration for selected software components such as copy config files templates if required. In the same go `RTE_Components.h` file is generated. At build time these tools are required to set `_RTE_` define to indicate that.

Some components have been written to allow (easier) usage outside of an "RTE enabled system". In this case alternative approaches to figure out configuration are used. In [os_systick.c](https://github.com/ARM-software/CMSIS_6/blob/main/CMSIS/RTOS2/Source/os_systick.c#L28C1-L28C28) it always included `RTE_Components.h` just for sake of knowing which `CMSIS_device_header` to be included. To use this component outside of an "RTE enabled system" one would need to provide `RTE_Components.h` containing `#define CMSIS_device_header "my_device_header.h"`. We could of course mask `#include "RTE_Components.h"` by `#ifdef _RTE_` and allow providing `CMSIS_device_header` via an alternative way.

ReinhardKeil, 2024-11-18:
Additional information is now in this documentation https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-overview.md#_rte_-preprocessor-symbol

closed 2024-11-18 by ReinhardKeil