ARM-software/CMSIS_6 #215: core_ca.h defines for ISR and SGIR_INTID

NVMarkovich, 2025-01-24:
1. In core_ca.h, lines 644; 647; 650, the ISR_A_Pos, ISR_I_Pos, ISR_F_Pos should be not 13U, 12U, 11U, but 8U, 7U, 6U respectively, as per structure at lines 636-638.
2. In the same file, line 921, GICDistributor_SGIR_INTID_Msk should be not 0x7U, but 0xFU, as the INTID is 4 bits.

JonatanAntoni, 2025-01-24:
@NVMarkovich, thanks for raising this. Would you mind to raise a pull-request contributing the requested changes?

NVMarkovich, 2025-01-24:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe. Never provide your password!
> VORSICHT: Diese E-Mail kommt von einem externen Absender. Bitte keine Links anklicken oder Anlagen öffnen falls Sie den Absender nicht kennen. Niemals Ihr Kennwort eingeben!

Hello, I am a bit new at GitHub, and I thought it will be enough to raise the question and the colleagues from ARM will correct their file.

Hilscher Development & Test Center EOOD  |  Business Park Varna - B8, 402  |  9009 Varna | Bulgaria |  www.hilscher.com<http://www.hilscher.com/>

Important Information:
This e-mail message including its attachments contains confidential and legally protected information solely intended for the addressee. If you are not the intended addressee of this message, please contact the addresser immediately and delete this message including its attachments. The unauthorized dissemination, copying and change of this e-mail are strictly forbidden. The addresser shall not be liable for the content of such changed e-mails.

????????:
???? ?????????? ????????? ? ??????????? ??? ???? ??????? ???????? ?????????? ????????????? ?????????? ?? ???????? ????? ??????????. ??? ?????? ????? ?? ? ? ??????? ?? ?????????? ???? ????????? ?? ???????? ???? ????????? ? ??????????? ??? ???? ??????? ???? ???????????? ????????? ????????. ????????, ??????? ??? ????????? ? ????? ???? ?? ???????????? ??????????? ? ???? ????????? ??? ????????? ???????? ?? ?????? ? ?????? ?????????.

JonatanAntoni, 2025-01-24:
@NVMarkovich, the files you are referring to are community contributed.

NVMarkovich, 2025-01-24:
ok, when I finish reviewing the whole file, I will consider raising a pull-request. Meanwhile I plan to add the further new-found issues in the initial message, if You don't mind.