ARM-software/CMSIS_6 #230: Wrong SCB VTOR TBLOFF Mask

by-gnome, 2025-04-22:
In core_cmX.h files SCB_VTOR_TBLOFF_Msk is defined incorrectly.
For example, in core_cm4.h:
```
#define SCB_VTOR_TBLOFF_Pos 7U /*!< SCB VTOR: TBLOFF Position */
#define SCB_VTOR_TBLOFF_Msk (0x1FFFFFFUL << SCB_VTOR_TBLOFF_Pos) /*!< SCB VTOR: TBLOFF Mask */
```
Thus SCB_VTOR_TBLOFF_Msk is defined as 0xFFFFFF80, but it looks like it should be 0x3FFFFE00.
According to PM0214 STM32 Cortex®-M4 MCUs and MPUs programming manual:
```
Bits 31:30 Reserved, must be kept cleared
Bits 29:9 TBLOFF: Vector table base offset field.
It contains bits [29:9] of the offset of the table base from memory address 0x00000000. When
setting TBLOFF, you must align the offset to the number of exception entries in the vector
table. The minimum alignment is 128 words. Table alignment requirements mean that bits[8:0]
of the table offset are always zero.
Bit 29 determines whether the vector table is in the code or SRAM memory region.
0: Code
1: SRAM
Note: Bit 29 is sometimes called the TBLBASE bit.
Bits 8:0 Reserved, must be kept cleared
```

JonatanAntoni, 2025-04-23:
Hi @by-gnome,

This is an implementation defined behaviour for STM in your case. In CMSIS we implement the Arm Architecture and some Cortex implementation specifics. The Vector Table Offset Register `VTOR` is specified in the [Armv7-M Architecture Reference Manual (DDI0403E), section B3.2.5](https://developer.arm.com/documentation/ddi0403/ee):

[image]

Implementors (such as ST) are free to restrict implementation of `TBLOFF` to bits `29:7` (matching their devices memory layout) as well as put additional alignment requirements (512 bytes in your case). We can't recognize such implementation details in our implementation-agnostic library.

Hope this answers your question.

Cheers,
Jonatan

by-gnome, 2025-04-23:
Thank you, I got the answer.
Closing the issue.

closed 2025-04-23 by by-gnome