ARM-software/CMSIS_6 #235: [startup.c] issue when using with C++ compiler

tarek-bochkati, 2025-04-27:
Simply due to name mangling, when using a C++ compiler
the __PROGRAM_START could not be found at link stage.

thus we need to change this line below to be C++ proof
https://github.com/ARM-software/CMSIS_6/blob/cc94f668add0d7cda2b7fbd75000ad2a23f100c6/CMSIS/Core/Template/Device_M/Source/startup_Device.c#L40

The proposed change is the following
```C
#ifdef __cplusplus
extern "C" {
#endif

extern __NO_RETURN void __PROGRAM_START(void);

#ifdef __cplusplus
}
#endif
```

JonatanAntoni, 2025-04-28:
Hi @tarek-bochkati,

The assumption in CMSIS is that files are compiled based on their extension. I.e., `.c` files shall be compiled in C-mode and not in C+-mode. Header files are expected to be C/C++ interwork compatible. Compiling `.c` files in C++-mode might cause subsequent linking issues even if we add the "extern C" declaration in this place.

tarek-bochkati, 2025-05-14:
thank you @JonatanAntoni

closed 2025-05-14 by tarek-bochkati