ARM-software/CMSIS_6 #236: [startup.c] C++ compilation error due to `__NO_RETURN` used with `__PROGRAM_START` prototype

tarek-bochkati, 2025-04-27:
When using IAR C++ compiler, I get the following error for this line (fix applied according to #235 )

```C
#ifdef __cplusplus
extern "C" {
#endif

extern __NO_RETURN void __PROGRAM_START(void);

#ifdef __cplusplus
}
#endif
```

```
Error[Pe1866]: attribute does not apply to any entity	
```

I have checked that __NO_RETURN have been resolved to [[noreturn]] according to my compiler version and `cmsis_iccarm.h` content
https://github.com/ARM-software/CMSIS_6/blob/cc94f668add0d7cda2b7fbd75000ad2a23f100c6/CMSIS/Core/Include/cmsis_iccarm.h#L57-L65

I don't understand fully the error behind this but just changing the prototype to
```C
extern void __PROGRAM_START(void) __attribute__((__noreturn__));
``` 

made it work like a charm

do you have any hint about the underlygin problem ?

JonatanAntoni, 2025-04-28:
Hi @tarek-bochkati,

This relates to my answer to #235. In CMSIS we expect `.c` files to be compiled in C-mode, not C++-mode. In this case you are getting a confusion between C and C++. Compiling in C++11-mode the IAR header sets `__NO_RETURN` to the C++11 (and later) standard flag `[[noreturn]]`. But as this shall be a C declaration using the C++ declaration does not work. Unfortunately, the preprocessor resolving the defines doesn't know which parts are ment to be C++ and which are "extern C".

tarek-bochkati, 2025-05-14:
thank you @JonatanAntoni

closed 2025-05-14 by tarek-bochkati